ZoneMinder/zoneminder #4229: When using RtspToWEB, the stream does not stop after a long period of inactivity.

IgorA100, 2025-02-25:
If there is a long period of inactivity, a modal window appears

> **Are you still watching?**
> Video paused. Continue watching?

But stream does not stop when using RtspToWEB.

connortechnology, 2025-02-25:
Maybe.  The entire purpose is to stop bandwidth use... so....

IgorA100, 2025-02-25:
Ok.

connortechnology, 2025-02-26:
Worse yet, leaving the AYSW popup up all night results in chrome using 400% cpu...

IgorA100, 2025-02-26:
> using 400% cpu..

WOW!
I'll definitely watch it!

IgorA100, 2025-02-27:
I analyzed the code.
When idle and a modal window appeared, the pause command was executed every 10 seconds. This is not correct. In theory, it should not directly affect the increased CPU load, but it could lead to some failures, which in turn could increase the CPU load.
Also, now the thread stops and starts again.
I hope the problem will be solved.

PR #4234