ZoneMinder/zoneminder #4242: 1.36.35 floods logs with "non increasing dts, fixing." warnings

onlyjob, 2025-03-02:
Zoneminder 1.36.35 (on Debian) floods logs with high rate of specific warnings:

```
WAR [zmc_m22] [non increasing dts, fixing. our dts 964134450 stream 0 last_dts 964228770 difference 94320 last_duration 4500. reorder_queue_size=0]
```

(`dts {number} stream 0 last_dts {number}` vary but other parts of the message remain constant.)

This appears to be a regression that manifested after upgrade from 1.36.33. Please advise.

connortechnology, 2025-03-02:
Known problem, lots of discussion in forums about it.  release-1.36 branch may have a fix.  Would appreciate your testing of release-1.36 branch as this is what will go into 1.36.36.  Other user who was experiencing the issue just reverted to .33.

onlyjob, 2026-02-01:
I had to suppress those warnings as follows, in `/etc/systemd/system/zoneminder.service.d/override.conf`:

```
[Service]

## https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#LogFilterPatterns=
## https://github.com/ZoneMinder/zoneminder/issues/4242
LogFilterPatterns=~(non increasing dts)
```

closed 2026-11-16 by connortechnology