ZoneMinder/zoneminder #4270: Changing the translation prevents Mass-Unarchive button from working

locutus13, 2025-03-19:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.35 - 
- How you installed ZoneMinder: ppa:iconnor/zoneminder-1.36
- Full name and version of OS: Ubuntu 22.04.5 LTS
- Browser name and version: Firefox 136.0.2

**Describe the bug**
Changing the translation (e.g. to German) prevents mass-"Unarchive" button in the list view from working

**To Reproduce**
Steps to reproduce the behavior:
1. Options - Users
2. Change the language for a user and/or the system to something that has a translation for the text "Archived"
3. try to unarchive any event in the list view - the button on the top will be greyed out and non-functional
4. You can still archive events with the Archive button and their status will show as a translation of "Archived" (which may be the reason why it stops working)

**Expected behavior**
The Mass-Unarchive button in the list view should be unaffected by translations

IgorA100, 2025-03-31:
It's hard to answer anything about version 1.36.XX, since 1.38 is expected to be released very soon
The design of 1.36 and 1.38 is very different.
If only **Isaac** could clarify something.

connortechnology, 2025-03-31:
Please change to ppa:iconnor/proposed to test this fix. Or apply yourself manually.

commit fcacb049 (2025-03-31): Use translated yes instead of 'yes'.  Fixes #4270
commit db6b69af (2025-03-31): Use translated yes instead of 'yes'.  Fixes #4270
closed 2025-03-31 by connortechnology