ZoneMinder/zoneminder #4273: boken images look bad

connortechnology, 2025-03-25:
Say when we go from montage to console by click on the word console.  We kill all the streams, which is good because otherwise it can take a while for the browser to actually stop streaming itself and go to the console.

However for a brief second we see a bunch of broken images before the page changes.

Maybe when stopping these streams we should replace them with a 1 pixel black image or something instead.

IgorA100, 2025-03-25:
Ohhhh...
My Montage page has been changed so much that I don't even know what to answer.... :(
I'm already losing hope for my PR to be approved

> However for a brief second we see a bunch of broken images before the page changes.

Yes, I just have a static image displayed for a split second. Is that bad?
I don't quite understand why a static image needs to be changed to 1 pixel.
Let's say we change it to 1 pixel. What should we do with the image grid? What should we do with the "Monitor status position" labels?
You can just make a white sheet, but why?
I don't understand anything :(

connortechnology, 2025-03-25:
I meant that the 1px image but stretched to whatever size.  Just to save ram or whatever.  
Or maybe we actually hide the img or even all montage content entirely while the page transition is happening.

IgorA100, 2025-03-25:
> I meant that the 1px image but stretched to whatever size. 

I'm aware.

> Just to save ram

I haven't analyzed this point in depth, but the browser itself should correctly clear memory. And I'm 99% sure that we won't save memory..... When I was implementing type MSE, there were moments with clearing memory. It would still be cleared, but with a delay. With large buffer sizes, this would be bad, because there could be a short-term memory shortage. The situation is different here.

> or whatever.

For example? Nothing comes to mind.

> Or maybe we actually hide the img or even all montage content entirely while the page transition is happening.

As an option...

IgorA100, 2025-03-25:
Okay, I'll think about it.

IgorA100, 2025-03-27:
I tried to make a smooth hiding of all content using JQ "animate".
In principle, it works out well.
I have a lot of PRs that have not been approved yet, so I will no longer be able to create new PRs, because it will be a complete mess due to the fact that I do not know which of my PRs will be approved and which will not.

closed 2025-04-02 by connortechnology