ZoneMinder/zoneminder #4278: ZM seems to overload a 6-core 16GB ram.

guillaumesoucy94, 2025-03-31:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
V1.36.35

- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
Using PPA

- Full name and version of OS
Ubuntu Server 24.04

- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
A clear and concise description of what the bug is.

I do have 20 cameras, all in monitor mode, and their resolution are all set to 640x480 (was higher before I reduced for testing purposes). The dedicated host could run for a few hours with a load of 2~3 before skyrocketing to something like 30, then on the attached consol (LCD panel from the crash cart), it shows a message about the host is out of memory and at every occurrence if I wait 10~15 minutes the host is returned to a load of 2~3 with RAM usage of about 7~8GB

**To Reproduce**
Steps to reproduce the behavior:
1. I set up my ZM host
2. Hooked 20 remote cameras to it.
3. Wait a few moments. 3~4 hours.
4. Load is skyrocketing to 30.

**Expected behavior**
A fluidly working ZM system.

**Debug Logs**
```
Can't access the host right now as it is overloaded again, and I'm out of town. Just post this to have a chance to get a solution earlier than if I wait to get back home. I will post it ASAP.

```
Thanks in advance for your help 😊 

Guillaume

guillaumesoucy94, 2025-03-31:
Good morning,

This is what the load was this morning:

[image]

Right now, it is decreasing, but it jumps to 32+, making the system stuck due to high load.

Guillaume

connortechnology, 2025-03-31:
Unfortunately I find that load is not a very good metric for cctv systems.  I have systems where load is over 100, but cpu use is 30%. 

The running out of memory is the bigger problem.  You did use the word "remote". Do you mean the remote monitor type?  Try using the ffmpeg type instead.

guillaumesoucy94, 2025-04-04:
I tried to switch one camera to try it out from remote to ffmpeg but it refuse to connect. The status says "Not  running" in red.

I entered the camera address, port and the path, including "user:password", just like in remote but in "Source Path" field while being in ffmpeg.

guillaumesoucy94, 2025-04-10:
Hello,

I don't want to rush anybody here, but I'm looking for a solution to the issue. 🙂

Thanks

Guillaume

guillaumesoucy94, 2025-04-11:
Also, how to use ZM with ffmpeg?

guillaumesoucy94, 2025-04-11:
Also, how do you use ZM with ffmpeg?

guillaumesoucy94, 2025-04-26:
I was able to find out.

Performances are way better now.

guillaumesoucy94, 2025-05-02:
I was able to find out how to correctly formulate the syntax the URL to put in the field when using fmpeg. (I was asking how to do so in my previous message, but I figured it out by myself)

Using ffmpeg instead of remote "fixed" the slowness issues (for me). 

If someone had slowness issues with ZoneMinder, please give a try to ffmpeg, as it seems less resources intensive, and now, my ZoneMinder setup is working perfectly fine. I can now enjoy it. 

Thanks to all for helping to figure out what was wrong. 😀

Guillaume