ZoneMinder/zoneminder #4316: Montage Layout spacing

connortechnology, 2025-04-23:
At 24 wide and 48 wide the spacing between grid-item-content is noticeable and annoying.
Appears to be an inline style inset: 0 1px; Being an inline style it is hard to override, perhaps there is a grid stack setting for it?

IgorA100, 2025-04-30:
@connortechnology
Have you tried adding "24 Wide" and "48 Wide" ?
In "montage.js" -> "initGridStack" there are options:
```
margin: '0 1px 0 1px',
cellHeight: '4px', //Required for correct use of objGridStack.resizeToContent
```
cellHeight is not recommended to be less than 4px yet, otherwise there may be performance issues! In the new version of GridStack this problem may have been fixed, but I haven't tested it yet :(

connortechnology, 2025-05-05:
margin: '0' worked.

closed 2025-05-05 by connortechnology