ZoneMinder/zoneminder #4319: REGRESSION:  Back to an event list from an event requires a manual refresh now.

Simpler1, 2025-04-30:
**Describe Your Environment**
- Version of ZoneMinder: 1.37.66 (latest commit: b357ce52aa6397818ff1d8c362122fbc8f6e3df5)
- How you installed ZoneMinder: Docker: zoneminder-containers/zoneminder-base with eventserver-base
- Full name and version of OS: Linux Mint 20.3 Cinnamon
- Browser name and version: Chromium: Version 130.0.6690.0 (Official Build) stable (64-bit)

**Describe the bug**
Some change was made between the following two commits: (It's been a month and a half since I've updated)
cd914f885dda2dee5675401e065a6cb791d3267e on March 3, 2025
b357ce52aa6397818ff1d8c362122fbc8f6e3df5 on April 22, 2025
When clicking on an event from a list of events and then clicking the back button (in the top left of the UI), the list of events would always display.  Now, it just shows the header with no list.  I have to manually refresh the page (F5) to get the list to reappear.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any list of events
2. Click on an event
3. Click on the left arrow in the top left of the UI
4. Notice that the list is not displayed
5. Manually refresh the page by pressing F5
6. Notice that the list appears.

**Expected behavior**
A manual refresh should not be required after backing out of an event.

@IgorA100

IgorA100, 2025-04-30:
Chromium 109.0.5414.120 & Win 8.1 I can't reproduce the problem.
As far as I remember, I didn't have any problems with the newer Chromium on Win 10 either.
Maybe the problem is only in Linux? or in some ZM settings.
You don't have a chance to check in Windows?
I don't have any computers with Linux at the moment :(

Simpler1, 2025-04-30:
I've verified the problem on Android 14 with Chrome also.   The problem
doesn't occur on Android 14 with Firefox.

IgorA100, 2025-04-30:
Android 10 & Chrome 135.0.07049 no problem found :(
Probably the problem is related to caching. But the strange thing is that everything is read from the cache except DataTable.
I need to somehow reproduce this problem on my own....

IgorA100, 2025-04-30:
Can I get a screenshot including console?

IgorA100, 2025-04-30:
@Simpler1 
Try to remove the function "window.addEventListener('beforeunload', function(event) {" in the skin.js file
Will your problem remain or will everything be fine?

Simpler1, 2025-04-30:
Removing that function seems to have fixed the problem.

IgorA100, 2025-04-30:
@Simpler1 
Since I can't reproduce the problem, I ask for a little more help.
1. Bring back the previously deleted function.
2. Comment out the line "**mainContentJ.css({display: "none"});**" in this function and check.
3. If p.2 doesn't help, then comment out the line "**mainContentJ.css({flex: "0 1 auto"});**"
Report the result.

Simpler1, 2025-04-30:
If I comment this entire function then the problem does not occur:
`mainContentJ.animate({height: 0}, 300, function rollupBeforeunloadPage() {`

IgorA100, 2025-04-30:
Thank you!
But this is very, very strange....
I will try to figure out the reason.
It's too bad that I still can't reproduce the problem :(.

IgorA100, 2025-04-30:
Android 15, Chrome 135.0.7049.111 the problem does not reproduce :(

Simpler1, 2025-04-30:
Reproducible on Android 14 with Chrome 135.0.7049.100
These are the latest available for my Samsung Galaxy S22

IgorA100, 2025-04-30:
I don't have Android 14 :(
Perhaps some ZM settings affect the problem... ??
Try in the file "**\skins\classic\views\js\event.js**" in the function `"bindButton('#backBtn', 'click', null, function onBackClick(evt) {"` comment out the line `"evt.preventDefault();"` I don't think this will fix the problem. But.....
Also try in the function **"window.addEventListener('beforeunload', function addListenerGlobalBeforeunload(event) {"** in the skin.js file replace **"mainContentJ.css({display: "none"});"** with **"mainContentJ.css({display: "none", height: "", flex: ""});"**

IgorA100, 2025-12-02:
It's possible this issue only occurs on Android 14 with Chrome or Chromium.
I can't reproduce the problem on any devices.

connortechnology, 2025-12-02:
I'm on Android 16, Chrome 142.0.7444.171

connortechnology, 2025-12-02:
My wife's phone is older I can test with that.

Simpler1, 2025-12-02:
My Samsung Galaxy S22 phone has since updated to Android 16 and is running Chrome 142.0.7444.171.
The problem still exists with the latest.

Simpler1, 2025-12-05:
Replacing that code does not change the behavior.

IgorA100, 2025-12-05:
> not change the behavior.

This is very strange. I need to reproduce the problem myself, but I don't know how to do it :(

Simpler1, 2025-12-05:
> Try changing two functions and reporting the console readings:
> 
Clicking on the back button from an event to events:
>     1. Before pressing any "Leave" or "Cancel" button
> 
[image]

>     2. After pressing "Leave"
> 
[image]

>     3. After pressing "Cancel"
> 
[image]

The problem does not happen with this code (even after clicking Leave).

IgorA100, 2025-12-05:
> The problem does not happen with this code

I see that beforeunload is called once, and the page wasn't restored from bfcache. Yes, everything should work correctly.
If you make this change: `//event.preventDefault();`, will the problem reoccur?

Simpler1, 2025-12-05:
> > The problem does not happen with this code
> 
> I see that beforeunload is called once, and the page wasn't restored from bfcache. Yes, everything should work correctly. If you make this change: `//event.preventDefault();`, will the problem reoccur?

if `event.preventDefault();` is commented, the problem returns.

IgorA100, 2025-12-06:
Try this code (without using JQ):
```
  function addListenerGlobalBeforeunload(event) { 
    window.removeEventListener('beforeunload', addListenerGlobalBeforeunload);
    console.log("event.target.activeElement=>", event.srcElement.activeElement);
    console.log("event.target.activeElement.href=>", event.target.activeElement.href);
    console.log("event.target.activeElement.outerHTML=>", event.target.activeElement.outerHTML);
    if (navbar_type == 'left') {
      closeMbExtruder(updateCookie = true);
    }

    if (mainContent) {
      var animation = mainContent.animate([
        {transform: 'translateY(-'+window.outerHeight+'px)'}
      ], 300);
      animation.addEventListener('finish', function() {
        mainContent.style.transform = 'translateY(-'+window.outerHeight+'px)'};
      });
    }
  }
  window.addEventListener('beforeunload', addListenerGlobalBeforeunload);
```

Simpler1, 2025-12-07:
> Try running this code: And understand, right now it's not you who's helping me, but I'm helping you.
> 
> ```
>   window.addEventListener('pageshow', (event) => {
>     console.log("event_pageshow_persisted=>", event.persisted, mainContentJ[0].clientHeight);
>     //if (event.persisted) {
>       // Do any checks and updates to the page
>       if (mainContentJ[0].clientHeight < 1) {
>         window.location.reload( true );
>       }
>     //}
>   });
> 
>   function addListenerGlobalBeforeunload(event) { 
>     //window.removeEventListener('beforeunload', addListenerGlobalBeforeunload);
>     //event.preventDefault();
>     if (navbar_type == 'left') {
>       closeMbExtruder(updateCookie = true);
>     }
> 
>     if (mainContentJ) {
>       if (mainContentJ.css('display') == 'flex') {
>         // If flex-grow is set to a value > 0 then "height" will be ignored!
>         mainContentJ.css({flex: "0 1 auto"});
>       }
> 
>       mainContentJ.animate({height: 0}, 300, function rollupBeforeunloadPage() {
>         const btnCollapse = $j('body').find('#btn-collapseQQ');
>         if (btnCollapse) btnCollapse.css({display: "none"});
>         mainContentJ.css({display: "none"});
>       });
>     }
>     //event.returnValue = '';
>   }
>   window.addEventListener('beforeunload', addListenerGlobalBeforeunload);
> ```

Still broken.

[image]

Simpler1, 2025-12-07:
> > Still broken.
> 
> Please provide a screenshot. I understand that the buttons above the table are visible, but the table itself with the list of events is not?

[image]

IgorA100, 2025-12-09:
There won't be anything interesting in the browser console.
All logs should be written to the database.
Perhaps you have PHP & Java logging disabled in Options -> Logging?

[image]
[image]

connortechnology, 2025-12-09:
Do you have any plugins active?  That is my suspicion.  But this is on a phone right?  So can you have plugins on a phone?  I've never tried.  Maybe it is is something that comes with Samsung.

Long story short, we may just have to do a dirty workaround.

IgorA100, 2025-12-10:
Change the code to this and repeating "Go to the Events page -> Event page -> return to the Events page."
And republish the new logs.

```
  function writeToLog(logData) {
    ajax = $j.ajax({
      url: thisUrl + '?view=request',
      type: 'POST',
      data: logData,
      timeout: 0,
      success: function(data) {
      },
      error: function(jqxhr) {
      }
    });
  }
  
  // https://web.dev/articles/bfcache Firefox has a peculiar behavior of caching the previous page.
  window.addEventListener('pageshow', (event) => {
    writeToLog(
      {
        request: 'log',
        task: 'create',
        level: 'INF',
        message: 
          "pageshow_event_persisted=>" + event.persisted + "<>" + mainContentJ[0].clientHeight + "<=|||" +
          "pageshow_window.location=>" + window.location + "<=|||"
          ,
        file: "Beforeunload"
      }
    );
    if (event.persisted) {
      // Do any checks and updates to the page
      writeToLog(
        {
          request: 'log',
          task: 'create',
          level: 'INF',
          message: "!!!pageshow_event_persisted_reloadPage!!!",
          file: "Beforeunload"
        }
      );
      window.location.reload( true );
    }
  });

  function addListenerGlobalBeforeunload(event) { 
    //window.removeEventListener('beforeunload', addListenerGlobalBeforeunload);
    //event.preventDefault();
    if (navbar_type == 'left') {
      closeMbExtruder(updateCookie = true);
    }

    writeToLog(
      {
        request: 'log',
        task: 'create',
        level: 'INF',
        message: 
          "bul_event.target.activeElement.href=>" + event.target.activeElement.href + "<=|||" +
          "bul_event.target.activeElement.outerHTML=>"+ ((event.srcElement.activeElement.tagName=='BODY') ? "<BODY>" : event.srcElement.activeElement.outerHTML) + "<=|||" +
          "bul_window.location=>" + window.location + "<=|||"
          ,
        file: "Beforeunload"
      }
    );

    if (mainContentJ) {
      if (mainContentJ.css('display') == 'flex') {
        // If flex-grow is set to a value > 0 then "height" will be ignored!
        mainContentJ.css({flex: "0 1 auto"});
      }

      mainContentJ.animate({height: 0}, 300, function rollupBeforeunloadPage() {
        const btnCollapse = $j('body').find('#btn-collapseQQ');
        if (btnCollapse) btnCollapse.css({display: "none"});
        mainContentJ.css({display: "none"});
      });
    }
    //event.returnValue = '';
  }
  window.addEventListener('beforeunload', addListenerGlobalBeforeunload);
```

Simpler1, 2025-12-11:
[image]

The tabled refreshed automatically with this code when returning to the events page.

IgorA100, 2025-12-11:
Please check PR #4480

IgorA100, 2025-12-11:
I'd be interested in seeing a video of a new problem.
The problem is "bfcache." For some reason, the browser is retrieving "broken data" from the cache. Try disabling "bfcache" in the browser. Not just disabling caching, but specifically "bfcache."
To solve this problem, I need to think and experiment. But there are two problems:
- I can't reproduce the problem yet. I need to install Linux + Chromium somewhere and try.
- I don't have free time for experiments.