ZoneMinder/zoneminder #4323: Future: Support for languages ​​other than English in camera names, tags, etc.

IgorA100, 2025-05-01:


connortechnology, 2025-05-02:
Theonly issue is we create a symlink with the name of the monitor in the FS, so we are limited to what characters are supported by the fs.   But we don't need to do that, and could optionally use a stripped down version of the actual name.  

What is stopping us from using other languages in tags?  I'm not aware of anything...

IgorA100, 2025-05-03:
I tried to figure it out once, but put it off...
There were some problems with storing data in SQL

IgorA100, 2025-05-03:
Here is an example of an error when adding a tag:
`ERR [SQL-ERR dbFetchAll no result, statement was 'INSERT INTO Tags (Name, CreatedBy) VALUES (?, ?)'params: \xd0\xba\xd0\xbb\xd0\xb0\xd1\x86,2], referer: http://v.хх/zm/index.php?view=event&eid=1155360&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=Group&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Battr%5D=Monitor&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bval%5D=&filter%5BQuery%5D%5Bterms%5D%5B2%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B2%5D%5Battr%5D=StartDateTime&filter%5BQuery%5D%5Bterms%5D%5B2%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B2%5D%5Bval%5D=2025-05-03+11%3A23%3A50&filter%5BQuery%5D%5Bterms%5D%5B3%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B3%5D%5Battr%5D=EndDateTime&filter%5BQuery%5D%5Bterms%5D%5B3%5D%5Bop%5D=%3C%3D&filter%5BQuery%5D%5Bterms%5D%5B3%5D%5Bval%5D=&filter%5BQuery%5D%5Bterms%5D%5B4%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B4%5D%5Battr%5D=Notes&filter%5BQuery%5D%5Bterms%5D%5B4%5D%5Bop%5D=LIKE&filter%5BQuery%5D%5Bterms%5D%5B4%5D%5Bval%5D=&filter%5BQuery%5D%5Bterms%5D%5B5%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B5%5D%5Battr%5D=Tags&filter%5BQuery%5D%5Bterms%5D%5B5%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B5%5D%5Bval%5D=&filter%5BQuery%5D%5Bterms%5D%5B6%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B6%5D%5Battr%5D=Archived&filter%5BQuery%5D%5Bterms%5D%5B6%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B6%5D%5Bval%5D=&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&sort_field=StartDateTime&sort_asc=1&page=1`

connortechnology, 2025-05-03:
Uh.... why is it using dbFetchAll to do an insert?  That's weird.  It may have to do with the database collation, mysql defaults to latin1, but we should be using utf8mb4.  I've had lots of troubles over the years as they figured all that out.  My db was utf8mb3 for a while.

IgorA100, 2025-05-03:
> Uh.... why is it using dbFetchAll to do an insert?

I don't know.
MySQL DB encoding is utf8mb4