ZoneMinder/zoneminder #4324: Not honoring Recording: RecordingSource: Secondary

juan-ferrer-toribio, 2025-05-02:
**Describe Your Environment**
- Version of ZoneMinder 1.37.66~20250430.288-bookworm
- How you installed ZoneMinder Deb package
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)

In the camera's Source tab, I set the _Source Path_ to a 720P & 6FPS stream and the _SecondarySourcePath_ to a 4K & 25FPS stream, both streams come from the same camera. This way, I'll use the primary stream for analysis and visualization (using less CPU), and the secondary stream for passthrough recording.

But when I set _Secondary_ as _RecordingSource_ in _Recording_ tab, Zoneminder ignores it and still uses primary _Source Path_ for recording.

connortechnology, 2025-06-20:
The second stream was only ever implemented as far as using it for a secondary AUDIO source.  

Nothing else has been implemented.

juan-ferrer-toribio, 2025-06-21:
What do you think about implementing it also as a secondary source for video?

I could help you with that.

connortechnology, 2025-06-21:
That is the plan.  Feel free to help, but I think the changes will be too much for 1.38, but again, 1.39 is right around the corner.