ZoneMinder/zoneminder #4330: Memory leak?

baudneo, 2025-05-21:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
03ecb8b5eae4557db72832cfca65f86539b0f3d2
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
` source`
- Full name and version of OS
 `ubuntu 22.04`
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
I have a filter running in the foreground that is moving all recordings from one drive to another, IDK if this is what is causing the issue or just a coincidence. I also upgraded to latest master commit today, so that may be the issue.

I have a reolink duo v2 that is eating 25.3GB of RAM, even though I have that monitor `Capturing` set to `None`. IDK why the zmc -m5 proc is still up, the monitor isnt capturing in the web UI. When monitor 5 is capturing, the RAM just keeps growing. When monitor 5 isnt capturing, the RAM stops growing, but doesnt free. 

Even after the long foreground filter has completed, the zmc -m5 proc shows in htop and doesnt free the mem. So, it seems it isnt related to the filter running?

Also, I notice that ZM uses far more resources from this commit than it did from May 11th (just FYI)

[image]
[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
Memory is freed

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2025-05-21:
The filter can crease issues accessing the events table which can then slow down anything accessing the events table.  

Um Capturing = None...  no it shouldn't be running.  Something very strange is happening.

baudneo, 2025-05-21:
After a few reboots, things seem ok again. It does seem like a long running foreground filter is what's causing the issue.

I'll keep an eye on things and update this issue if I can recreate.

baudneo, 2025-05-25:
Going to close this, it seems to be due to running a long filter in the foreground, which isnt really a thing people should/will be doing much.

closed 2025-05-25 by baudneo