ZoneMinder/zoneminder #4335: Performance regression?

baudneo, 2025-06-09:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
latest master head
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
`source`
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
ZM is using far more resources than it did last month. After May 11th, I noticed that zmc processes were using more CPU (and sometimes RAM). I've been running ZM for years and I have always been able to leave the Montage page open with 5 cameras running and it will run solid for days, no refresh or anything needed, just rock-solid camera streams.

Since building from HEAD 4 days ago, after about 4+ hours, the ZM montage tab will freeze and need to be closed. My CPU util was around 16-18% with my usual workload for the past few years, after May 11th it now sits at 25-40%. `htop` shows `zmc` processes as the CPU hogs.

I dont have any useful information besides describing what I am noticing and bringing it to attention.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2025-06-09:
THere have really not been very many changes made lately.  I see some stuff to do with the onscroll and Are You Still Watching bits.  Perhaps those are having unintended consequences.

baudneo, 2025-06-10:
I can try and take some time to debug. I really don't think it's a client issue, as my desktop isn't bogged down and is responsive, just the ZM tab ends up "frozen" and then chrome pops up the "page is stuck, wait or exit" form. 

The rtsp source is go2rtc, I have another NVR (frigate, for the wife approval factor) install running in another container that uses the go2rtc sources and when zm freezes, the other NVR shows no signs of stress or issues, it just keeps streaming from go2rtc without issue.

my largest resolution camera is the reolink dual lens and its zmc process is eating the most cpu (to be expected). Maybe it has something to do with very hi resolution streams? I was also getting some mem leak, like behavior last week on this monitor.

What would be the best way to debug? Just start the zmc process and take notes on cpu util? I'll also re-install some old packages and see if I can correlate a commit with the increased usage.

My homelab hasn't had any new software installed or new lxc, VMs, etc. so I don't think it's something like that, but am not ruling it out.

baudneo, 2025-06-11:
> In the past we have seen infinite loops that gradually chewed up resources, but most of those have long since been fixed.  Good news is that there aren't many commits to bisect.  
> 
> If it is the browser UI locking up, then that is where we have to look.  I would check the js console and the network tab looking for anything out of the ordinary. 

Kk, I'll take a look tonight. I think I looked in the console but didn't notice anything. But to be fair, I was doing 6 other things at once so wasn't focused. I will take some time this evening to try and get some info.

baudneo, 2025-06-19:
Things are better now, I have gone back and forth between commits from May until current and cant really nail anything down. Probably something strange going on, on my end.