ZoneMinder/zoneminder #4381: When using go2rtc, only Primary stream is always used and switching to Secondary does not work

IgorA100, 2025-07-23:
I think in the go2rtc config you can do it like this:
streams:
  "id_0":
    PATH for primary stream
  "id_1":
     PATH for secondary stream

Isaac, if you do this, I'll do the rest in JS.
If you agree, I will sketch out the draft code.

connortechnology, 2025-07-23:
Sure. We also need the ability to stream from ZM directly intead of hitting the camera.  So there might be 4 streams.

IgorA100, 2025-07-23:
Isaac, when you make changes in the go2rtc config as I wrote, you can check with #4382 streams should switch.

IgorA100, 2025-07-23:
> We also need the ability to stream from ZM directly intead of hitting the camera.

Do you mean viewing recorded events from a file using go2rtc or am I missing something????

connortechnology, 2025-07-23:
No, using the rtsp_server functionality of ZM to re-stream.  We may have to apply privacy zones or other changes to the video. Also many cameras can't handle too many streams. So re-streaming is a plus.

IgorA100, 2025-07-23:
> using the rtsp_server functionality of ZM to re-stream.

Yes, I understand.
By the way, you don't want to use go2trc to view recorded events, it can do it, right?

connortechnology, 2025-07-23:
I implemented the C++ side parts.

The stream choice doesn't seem to stick on a reload.

IgorA100, 2025-07-24:
> The stream choice doesn't seem to stick on a reload.

Yes, you are right, we do not save the selected stream.
We will not be able to save the selected stream in cookies, because the cookie size is limited. In case of using hundreds of cameras, there will be problems.
Perhaps we can save in PHP sessions, the memory limit is higher there, but I do not know how this will affect performance. The entire session is loaded into memory.
You can also save in the DB.

At the moment, it works like this:
For RTSP2Web, there is a "Stream source" setting. When opening the Watch page, the "Stream quality" selector is set in accordance with the "Stream source" value from the monitor settings.
Since we do not use the "Stream source" setting for go2rtc (and by default we always select the "Primary" stream), a collapse occurs.
We can also start using the "Stream source" setting from the monitor for go2rtc. In this case there will be no collapse, but the selected "Stream quality" selector will not be saved (as for RTSP2Web)
Isaac, I'm waiting for your decision.

IgorA100, 2025-07-24:
Although... it turns out we save "zmStreamQuality", but it probably doesn't work correctly....
I will propose a solution in the near future.

IgorA100, 2025-07-24:
@connortechnology 
Try this: #4383

closed 2025-07-25 by IgorA100