ZoneMinder/zoneminder #4415: When clicking the "DEBUG" button on the Filter page, the error "form is not defined" will appear if "ZM_ENABLE_CSRF_MAGIC" is disabled

IgorA100, 2025-08-09:
For a long time I couldn't understand where the GLOBAL variable "form" comes from. I couldn't find....
After a long debugging I realized that the global variable "form" appears due to the function "CsrfMagic.end" in "includes\csrf\csrf-magic.js"
At the same time, "CsrfMagic.end" will not be called if "ZM_ENABLE_CSRF_MAGIC" is disabled.

When you click the "DEBUG" button, the function "debugFilter" will be called
```
function debugFilter() {
getModal('filterdebug', $j(form).serialize());
}
```
Which tries to use the global variable "form", but it doesn't exist....

I think everything is wrong here....
You need to use "use strict"; for csrf-magic.js and clean things up. Also, in the debugFilter() function you should NOT use the global variable "form"

connortechnology, 2025-08-09:
I agree.

IgorA100, 2025-08-09:
I'm surprised that for so many years no one has pressed the "DEBUG" button with ENABLE_CSRF_MAGIC disabled :)))

connortechnology, 2025-08-09:
It really isn't recomended to disable csrf... should probably be hidden under and "Advanced" section

IgorA100, 2025-08-09:
> It really isn't recomended to disable csrf... 

Yes, of course. But I had some problem (I don't remember what it was) during the tests, I disabled CSRF and then forgot to enable it. Eventually I found the problem...

> should probably be hidden under and "Advanced" section

It should probably be in the "Security" section, but there doesn't seem to be such a section.