ZoneMinder/zoneminder #4418: ZMS process is not killed on Console page after cursor leaves monitor image

IgorA100, 2025-08-09:
On the Console page, hover the cursor over the camera image.
The image will enlarge and be viewed in real time, and the ZMS process will appear.
After the cursor is removed from the image, there is no video animation, but the ZMS process continues to exist and consume resources.

IgorA100, 2025-08-09:
Version 1.37.68~20250723155645-noble
Moreover, the version from half a year ago does not have this problem.

connortechnology, 2025-08-09:
This doesn't happen for me.

IgorA100, 2025-08-09:
> This doesn't happen for me.

Very strange....
Perhaps there is some difference in your and my build.
If I specify conkey for src, then the ZMS process is successfully deleted.

connortechnology, 2025-08-09:
There shouldn't be connKey for console, it is only to identify a socket to talk to zms with, so you can send commands and get status updates.

IgorA100, 2025-08-09:
You're probably right, but without conkey ZMS the processes multiply and aren't killed... I'll wait until you make a fresh build and check it on it. Maybe there won't be a problem.

IgorA100, 2025-08-09:
Hmm. Totally strange....
I'm now near another computer, and there is no problem without conkey. The process is killed. The browser on both machines is chomium.
Probably some kind of glitch with the browser, but the situation is not very good, I will monitor.

IgorA100, 2025-08-11:
On one computer, processes are still multiplying and not being destroyed.
I did not restart the browser.
I would really like to find the problem, but I do not know how to do it yet :(
This is a very bad problem, because it can simply hang the server.
Installing conkey does not help.
Clearing SRC does not help either.

[image]

connortechnology, 2025-08-11:
So what's the difference between the two machines?

I have seen something like this when using other ways of running cgi scripts like nginx does.  zms never gets a sigpipe.  SO it never knows to quit.  

Does the browser still know about it? Is the bandwidth count still increasing in the network tab?

connortechnology, 2025-08-11:
I see in skin.js  line 1136, we update src, but we aren't setting it ='' first.  Please try add a src=''; before setting it to the still value.

IgorA100, 2025-08-11:
Tried it - doesn't help.
I tried it the day before yesterday, didn't help either....
I also tried clearing the SRC value in the browser console, didn't help either.
After refreshing the page (F5 or Ctrl+F5 or Ctrl+R) the ZMS processes are not destroyed either.
Only restarting Apache on the server helps.

connortechnology, 2025-08-11:
Is it browser specific?  How about if you close the browser entirely?  

THe only work-around I can think of is to tell it to only do a certain number of frames, then die.  You can do this with frames= in the url.  Perhaps in js we can then refresh the url so it loads another set of frames if you continue to mouse over.  

Another option is to add a connKey and send it the quit command, but what if that fails?  We would need a keepalive as well.

IgorA100, 2025-08-11:
> Is it browser specific?

It seems not.
It appears on Firefox. Chromium, Chrome.
I noticed an interesting point... It appears only if you go to the server's IP address. If I go through the domain name - there are no problems. In case of going through the domain name on the way to the server there is a proxy reverse Nginx.
There are also no problems if I connect through OpenVPN.

> You can do this with frames= in the url.

Yes, it worked! The specified number of frames is sent and the ZMS process is killed. If you quickly point it many times, the ZMS processes multiply, but then they are all gradually killed. That is, the load on the server increases, but then everything stabilizes.

IgorA100, 2025-08-11:
> Another option is to add a connKey and send it the quit command, but what if that fails? We would need a keepalive as well.

I'll try this option later.

IgorA100, 2025-08-11:
> Another option is to add a connKey and send it the quit command, 

I couldn't do it quickly.
As I understand it, MonitorStream objects on the Console page don't exist? I need to include the MonitorStream.js file and create new MonitorStream objects and then execute CMD_PLAY and CMD_QUIT.
Right?

IgorA100, 2025-08-11:
> Really not a great option.

I agree, I don't really like it either.. It's a last resort if nothing else helps...

> frames= and if we can attach an event to the stream ending so it can restart it would be the most lightweight.

I'll think about it...

IgorA100, 2025-08-21:
I looked at the connection on the server, they are all in the "CLOSE_WAIT" state

[image]

While on the client, they are all in the "Fin Wait 2" state

[image]

I have read and thought and tried a lot, but nothing helps. Here is an example:

> Socket enters the CLOSE_WAIT state when the remote end terminates the connection sending a packet with the FIN flag set. It then waits in this state for the local application to close() the socket and then sends its own FIN to the client and transitions the socket to the LAST_ACK state. See also [the TCP state transition diagram](http://en.wikipedia.org/wiki/File:Tcp_state_diagram_fixed_new.svg) and [RFC 793](http://www.ietf.org/rfc/rfc793.txt).

https://serverfault.com/questions/160558/how-to-not-get-so-many-apache-close-wait-connections

It seems to me that ZMS should not rely only on receiving a SIGPIPE signal, but should also check the state of open sockets and close them if necessary. But I am not an expert in network solutions :(

connortechnology, 2025-08-21:
If the browser doesn't close the stream, there isn't much we can do.

This is actually why we do things like img.src=''; img.src=new_url.  I wonder what happens if we actually remove the img from the DOM?  We currently css transform and changing the src to do the popup stream, but what if we left the still image alone, and added a new streaming img to the dom on top and then deleted it on mouse out.

IgorA100, 2025-08-22:
> When SRC changes, the client probably sends FIN and waits for confirmation from the server, but cannot wait for a response and so on endlessly...

Yes, I think it is necessary to "catch" when a FIN "arrives" from the client and at that moment close the socket. I have read a lot more and this is the correct way to work with sockets.

connortechnology, 2025-08-22:
You don't.  zm_stream should not need to deal with this.  There is something messed up in your network.

IgorA100, 2026-08-20:
I'm closing this as it seems the problem has been resolved.

closed 2026-08-20 by IgorA100