ZoneMinder/zoneminder #4438: No more Analysis on http sources?

brkai, 2025-09-13:
Hi,
i'm wondering if anybody can confirm this:

Since several years i had my monitors setup with source "http" and was accessing the MJPEG streams of my older cameras and fetching snapshot images of the newer ones to save resources on my server.
Since a few days the los said "analisys at 0.00 fps" and no events were created any longer.
Currently i'm using v1.37.70 and also tried 2 releases earlier with no success.

Now the situation is:
I deleted and created all of the monitors from scratch, using ONVIF as ffmpeg source accessing the provided RTSP streams form the cams and everything is running quite OK so far.
Currently im fine with "my" solution....

Cheers
Kai

connortechnology, 2025-09-13:
I suspect the db upgrade just failed to convert the settings correctly.  There is no reason why http mjpeg would not do analysis.  

In 1.37 the Function (mocord, modect etc) was replaced with 3 more explicit settings Capturing, Analysing, Recording.  I suspect the Analysing just didn't get set to Always during the upgrade.

brkai, 2025-09-13:
Hi Isaak,
i'll chek and report ;-)