ZoneMinder/zoneminder #4446: Incomplete monitor deletion

zenonp, 2025-09-21:
ZM: 1.36.35-1.el9.x86_64

Deleting a monitor from the web interface does delete its storage, but it does not delete the symlink to it. Illustration: 

[image]

The directory "3" has been deleted, but the (now dead) symlink "Clone of lc1" has been left behind.

zenonp, 2025-10-03:
> Was there anything in the logs? Failure to delete it should have logged something.

I looked and saw nothing relevant in the logs. And if you look at the ownership and permissions of the symlink, there is no reason why a deletion should have failed. 

Unfortunately this was two weeks ago and my recollection of what I did before I deleted the monitor has faded away. My gut feeling is that it might be relevant, e.g. in case I renamed the monitor "Clone of lc1" before I deleted it.

connortechnology, 2025-10-03:
I have been able to reproduce it.  Should have a fix soon.

zenonp, 2025-10-03:
> I have been able to reproduce it. Should have a fix soon.

If you tell me how you reproduced it, I might remember what I did before I deleted the monitor.

connortechnology, 2025-10-03:
clone monitor, then delete it. 

The issue was in our system cmd to do the delete not taking into account the existence of spaces, and not reporting the error.

zenonp, 2025-10-03:
> The issue was in our system cmd to do the delete not taking into account the existence of spaces...

OK, that was an easy one. But I've got one even easier at #4445 :)